DigitalSecurity/btlejuice #1: Cannot register Service

TheClownDE, 2016-09-01:
Hi,

I cannot get it work!
I'm running Linux Min 17.2 in VirtualBox and I use one Dongle and the internal chip of my laptop.
But I get this error all the time:

btlejuice-proxy:
```
[i] Using interface hci0
[info] Server listening on port 8000
[info] Client connected
Configuring proxy ...
[status] Acquiring target e1:83:ce:44:3b:4d
[info] Proxy successfully connected to the real device
[info] Discovering services and characteristics ...
[status] Proxy configured and ready to relay !
```

btlejuice :
```
[i] Using proxy http://127.0.0.1:8000
[i] Using interface hci1
2016-09-01T11:44:48.684Z - info: successfully connected to proxy
2016-09-01T11:45:59.301Z - info: proxy set up and ready to use =)
2016-09-01T11:45:59.315Z - debug: start advertising
2016-09-01T11:45:59.321Z - error: cannot register services !
```

Any Ideas?
Thanks

virtualabs, 2016-09-06:
Hi,

did you turn off (or blacklist) the bluetooth service on the host machine ? This service is active on most Linux systems but interferes with the Bleno library. 

Check its status:

```
$ sudo bluetooth service
```

If service is active, stop it:

```
$ sudo service bluetooth stop
```

Be careful, stopping the service will shutdown all the BT interfaces so you must re-enable them right after (replace X by your interface number):

```
$ sudo hciconfig hciX up
```

Restart BtleJuice and you should be good to go.

virtualabs, 2016-09-07:
I've just noticed the following line:

```
[i] Using proxy http://127.0.0.1:8000
```

You cannot run *btlejuice* and *btlejuice-proxy* on the same machine, this is a current known limitation of some dependencies BtleJuice is using. You must launch *btlejuice-proxy* in a VM, and *btlejuice* on your host machine (providing the VM IP address in the options as the proxy IP).

TheClownDE, 2016-09-08:
Thanks!
Cloning my VM did the trick.
But I'm now facing a new problem:
I used Bleno to create a peripheral with one service (uuid: e015) and two characteristics (uuid: e016,  e017) but only the first characteristic shows up on after connecting.
When I connect without btlejuice both characteristic show up.
Any ideas why?

Thnaks for your help so far!

virtualabs, 2016-09-08:
What tool are you using to check the available characteristics (nRF Connect, ...) ? There is a known issue with *bleno* (a dependency used by *BtleJuice*) that causes some trouble.

Try to enable the *Keep handle values* option in the settings dialog before starting the proxy, it may solve your problem. This is an experimental option available in the last version of the tool, and it had been improved since then and will be more stable in a future release.

TheClownDE, 2016-09-09:
I'm using nRF Connect (Version 4.4.0 on Android 4.4.3) and when I connect directly I see both characteristics.

> enable the Keep handle values option in the settings dialog before starting the proxy
I'm using BtleJuice v 1.0.6 (which I expect to be the latest version) but the settings dialog from the WebGUI only show the "Reconnect on client disconnections" option!
What do you mean by "before starting the proxy"?
I need to start the proxy before I can start btlejuic and the WebGUI!?
BR,
TheClown

virtualabs, 2016-09-09:
My mistake, I thought I've published the latest version. Please upgrade BtleJuice to version 1.1.0 through npm, enable the *Keep handle values* in the settings dialog, and see if this solves your problem. 

The option is still marked as experimental because it has not been intensively tested yet (and also because it monkey-patches one of BtleJuice's dependency).

TheClownDE, 2016-09-09:
Thank a lot, that solved it!
One more minor think btlejuice -v still shows 1.0.6 but its definitely the new version (new settings option,...)  ;)

virtualabs, 2016-09-10:
Okay, I fixed the version. I'm closing this issue.

closed 2016-09-10 by virtualabs