DigitalSecurity/btlejuice #2: Hangs when acquiring target

rob-cowsley, 2016-09-06:
When selecting a target from the web interface the proxy seems to hang:

# btlejuice-proxy 

[i] Using interface hci0
[info] Server listening on port 8000
[info] Client connected
[i] Stopping current proxy.
Configuring proxy ...
[status]Â Acquiring target 00:0b:57:0c:61:c7

virtualabs, 2016-09-06:
There are many reasons, but the most common are:

1. Ensure your proxy machine (VM or other) has a bluetooth adapter attached to it and managed by the system (use *hciconfig* to check it out, you may paste the result here for further analysis)

2. Your bluetooth adapter is not enabled. Check its status with *hciconfig* and enable it if it was previously disabled.

rob-cowsley, 2016-09-06:
Still having the same issue, proxy adapter output below:

root@obsidian:~# hciconfig
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:19:0E:18:53:01  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:978 acl:0 sco:0 events:52 errors:0
	TX bytes:2477 acl:0 sco:0 commands:52 errors:0

hci1:	Type: BR/EDR  Bus: USB
	BD Address: 00:22:43:F8:75:85  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:511 acl:0 sco:0 events:23 errors:0
	TX bytes:339 acl:0 sco:0 commands:23 errors:0

root@obsidian:~# btlejuice-proxy 
[i] Using interface hci0
[info] Server listening on port 8000
[info] Client connected
[i] Stopping current proxy.
Configuring proxy ...
[status] Acquiring target 00:0b:57:0c:61:c7

rob-cowsley, 2016-09-07:
Additional info on the device:

root@obsidian:~# hciconfig hci0 features version revision
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:19:0E:18:53:01  ACL MTU: 1021:8  SCO MTU: 64:1
	Features page 0: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
		<3-slot packets> <5-slot packets> <encryption> <slot offset> 
		<timing accuracy> <role switch> <sniff mode> <RSSI> 
		<channel quality> <SCO link> <HV2 packets> <HV3 packets> 
		<u-law log> <A-law log> <CVSD> <paging scheme> <power control> 
		<transparent SCO> <broadcast encrypt> <EDR ACL 2 Mbps> 
		<EDR ACL 3 Mbps> <enhanced iscan> <interlaced iscan> 
		<interlaced pscan> <inquiry with RSSI> <extended SCO> 
		<EV4 packets> <EV5 packets> <AFH cap. slave> 
		<AFH class. slave> <LE support> <3-slot EDR ACL> 
		<5-slot EDR ACL> <sniff subrating> <pause encryption> 
		<AFH cap. master> <AFH class. master> <EDR eSCO 2 Mbps> 
		<EDR eSCO 3 Mbps> <3-slot EDR eSCO> <extended inquiry> 
		<LE and BR/EDR> <simple pairing> <encapsulated PDU> 
		<err. data report> <non-flush flag> <LSTO> <inquiry TX power> 
		<EPC> <extended features> 
	Features page 1: 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00
	HCI Version: 4.0 (0x6)  Revision: 0x1000
	LMP Version: 4.0 (0x6)  Subversion: 0x220e
	Manufacturer: Broadcom Corporation (15)
	Firmware 0.34 / 14

virtualabs, 2016-09-10:
Okay, can you describe exactly how you start both *btlejuice* and *btlejuice-proxy* and your machines setup please ? I need more info to try to solve this problem.

rob-cowsley, 2016-09-11:
I have two laptops connected via a single switch. The proxy is using the on board Bluetooth and the client is using an external Bluetooth adapter.

Output from both devices is below:

[client.txt](https://github.com/DigitalSecurity/btlejuice/files/466263/client.txt)
[proxy.txt](https://github.com/DigitalSecurity/btlejuice/files/466264/proxy.txt)

Is that output enough for you?

virtualabs, 2016-09-13:
I managed to reproduce this bug and found the root cause: *noble* library de-duplicates detected devices by default so they would not be detected again during target acquisition. 

I fixed the problem in the master branch but haven't made a new release yet. Is it possible for you to try this fix with a fresh install from the master branch (*git clone* then *sudo npm install -g*) or should I issue a new release through npm ?

virtualabs, 2016-09-15:
I released two more versions since then, I hope the latest (v1.1.3) will fix your problem ;). I tested it intensively with the same setup that allowed me to reproduce this behavior.

rob-cowsley, 2016-09-15:
Currently re-setting up a couple of machines as there was an issue with one of them.

Should be able to test and confirm today with any luck (assuming install works on a PI)

rob-cowsley, 2016-09-18:
Looks good. :) - may have another issue now with remote device disconnects, but will do some more testing before raising it.

closed 2016-09-18 by rob-cowsley

virtualabs, 2016-09-18:
Great, feel free to file a new issue ;)

dayasagar12, 2020-10-06:
The issue still persists. I am getting stuck at Acquiring target <Device MAC Address> line 

I am using 1.1.11 version and using 2 laptops with ubuntu VMs running on both. I run the proxy in one system and run btlejuice command in the another system where I can see the list of bluetooth devices, but I don't see output in localhost:8080 page once I get connected.