DigitalSecurity/btlejuice #3: Error to download during install

cloatre, 2016-09-08:
root@kali:~/BLE# npm install -g btlejuice
...
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v11-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@0.10.29 (node-v11 ABI) (falling back to source compile with node-gyp) 

-->the url does no longer exists

virtualabs, 2016-09-08:
> below more logs

I don't see any screenshot or further information about this issue. 

Anyway, you must install *BtleJuice* as a normal user and use *sudo* as stated in the documentation:

```
$ sudo npm install -g btlejuice
```

Trying to install as *root* will fail because of some permissions probles while installing some of the dependencies.

cloatre, 2016-09-09:
OK I have my kali with only the user Root, I will add another user and try again.

Below the screenshot that does not work in previous post:
[image]

virtualabs, 2016-09-09:
Seems your v8 version (Node.js) is too old to install the latest versions of Noble and Bleno. Upgrade Node.js (download and install the latest version) and it should be ok :)

cloatre, 2016-09-09:
Ok thanks, I have reinstalled noble and bleno, and update nodejs to 4.x and I still look to resolve some other errors:

[image]

[image]

virtualabs, 2016-09-10:
Check if the *nodejs* executable is installed on your system. If not, create a symlink from the *node* executable in */usr/bin*. See http://stackoverflow.com/questions/21168141/cannot-install-packages-using-node-package-manager-in-ubuntu for more details about this issue.

bik3te, 2016-09-19:
cloatre, it looks like a previous install issue.

1. Uninstall nodejs;
2. Remove every path including node_modules (/usr/local/lib or /usr/lib or both);
3. Reinstall the last verision of nodejs;
4. Update npm (sudo npm install npm -g);
5. Reinstall btlejuice.

This should be ok then..

cloatre, 2016-09-20:
Hello,

Ok thanks, I still have errors during installation but the proxy seems to work without errors.

I try to detect my bluetooth devices but I don't see anything an my client is disconnected when I clic on the web interface, see below:

[image]

bik3te, 2016-09-20:
Okay, I already had this issue on Ubuntu 16.04, use sudo in order to launch the proxy:
$ sudo btlejuice-proxy

If it can help, on my VM, service bluetooth is started and hciconfig hci0 is up.
On my host, service bluetooth is stopped and hciconfig hci0 is up.

cloatre, 2016-09-20:
Ok it seems that my version of bluetooth adapter is not good:

[image]

virtualabs, 2016-09-21:
Ok, issue closed.

closed 2016-09-21 by virtualabs