DigitalSecurity/btlejuice #7: Spoofing BT address

ColdHeat, 2016-09-26:
So I'm trying to pair my phone with a device through btlejuice and the phone does not pick up the device. 

I suspect this is due to the adapter having a different address. 

Apologies for making another ticket but is there a way to have btlejuice spoof the address of the victim device?

ColdHeat, 2016-09-26:
I did not realize spooftooph existed so not sure if this is worth including in btlejuice but will leave it up to you whether you want to close this or not.

virtualabs, 2016-09-26:
I usually use [bdaddr](https://ubuntuforums.org/archive/index.php/t-925611.html) combined with a CSR 4.0 BT USB dongle to spoof a bluetooth (MAC) address. 

Have you already paired your device with your phone ? Does it rely on BT bonding or just some kind of application-level pairing ?

virtualabs, 2017-06-28:
Considering it solved, issue closed.

closed 2017-06-28 by virtualabs