DigitalSecurity/btlejuice #8: Cannot find module 'bluetooth-hci-socket'

cn0xroot, 2016-12-17:
nice to talk with you.
when i do:

`sudo btlejuice-proxy`

```
module.js:472
    throw err;
    ^

Error: Cannot find module 'bluetooth-hci-socket'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/usr/local/lib/node_modules/btlejuice/node_modules/noble/lib/hci-socket/hci.js:6:26)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
```

can you help me to fix it? thans.

virtualabs, 2017-01-13:
Do you have gcc/g++ and all the required development libraries installed on your system ? If not, install the build-essential package (sudo apt-get install build-essential).

laughingMan, 2017-04-26:
I found that installing the "usb" npm package before installing the "btlejuice" package fixed this issue for me.

`npm install usb`

nikoc03, 2017-06-11:
Hello everyone, did you resolve this issue? I try to install bluetooth-hci-socket and I get the following:

> bluetooth-hci-socket@0.5.1 install /root/node_modules/bluetooth-hci-socket
> node-gyp rebuild
```

gyp ERR! configure error 
gyp ERR! stack Error: "pre" versions of node cannot be installed, use the --nodedir flag instead
gyp ERR! stack     at install (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:66:16)
gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/node-gyp.js:55:37)
gyp ERR! stack     at getNodeDir (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:77:20)
gyp ERR! stack     at PythonFinder.callback (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:44:7)
gyp ERR! stack     at PythonFinder.<anonymous> (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:470:14)
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:262:7)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:897:16)
gyp ERR! stack     at Socket.stream.socket.on (internal/child_process.js:340:11)
gyp ERR! System Linux 4.9.0-kali4-amd64
gyp ERR! command "/root/local/bin/node" "/root/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/bluetooth-hci-socket
gyp ERR! node -v v9.0.0-pre
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.
npm WARN You are using a pre-release version of node and things may not work as expected

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-hci-socket@0.5.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-06-11T15_42_36_544Z-debug.log
```
 
could someone help please?

cn0xroot, 2017-09-28:
Parrot Security OS work fine.

closed 2017-09-28 by cn0xroot

mike0x73, 2021-03-18:
I've found that the latest version of node just hates btlejuice so I wrote some scripts to get them installed. Hope this helps https://github.com/mike0x73/btlejuice-gattacker-install

Freecodercc, 2024-12-11:
@mike0x73 you are my hero！