DigitalSecurity/btlejuice #10: Cannot register services and no packet captured

cryptax, 2017-01-12:
I have a btlejuice proxy running on 127.0.0.1 port _9000_:

```
[i] Using interface hci0
[info] Server listening on port 9000
[info] Client connected
Configuring proxy ...
[status] Acquiring target xx:xx:xx:xx:xx:xx
[info] Proxy successfully connected to the real device
[info] Discovering services and characteristics ...
[status] Proxy configured and ready to relay !
```

I connect the btlejuice host to this proxy, start the web interface, and select my BLE device.
I get **`error: cannot register services!`** and of course **no packets captured**.
```
$ sudo btlejuice -u 127.0.0.1 -p 9000 -w
   ___ _   _       __        _          
  / __\ |_| | ___  \ \ _   _(_) ___ ___ 
 /__\// __| |/ _ \  \ \ | | | |/ __/ _ \
/ \/  \ |_| |  __/\_/ / |_| | | (_|  __/
\_____/\__|_|\___\___/ \__,_|_|\___\___|

[i] Using proxy http://127.0.0.1:9000
[i] Using interface hci0
2017-01-12T13:46:57.040Z - info: successfully connected to proxy
2017-01-12T13:47:12.251Z - info: proxy set up and ready to use =)
2017-01-12T13:47:12.256Z - debug: start advertising
2017-01-12T13:47:12.260Z - error: cannot register services !
```

NB. I run btlejuice with **sudo** because otherwise I get this warning:
```
bleno warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/bleno#running-on-linux
```

FYI, the btlejuice proxy is running inside a **docker** container. I checked that it has access to my BLE dongle (anyway it would have complained otherwise).

fractalx, 2017-01-12:
think to take a look about closed issues
your #1 fix your pb ;)

virtualabs, 2017-01-12:
Running BtleJuice proxy in a docker container does not work, this is why I took the VM approach rather than using docker. Docker does use the host's kernel HCI features but is not able to act as a Bluetooth Smart Central and Peripheral as the same time. You need two separate stacks (as far as I've understood), therefore two machines (but you can run one in a VM). 

It seems to be a limitation of the BT 4.0 kernel features, I've heard it works pretty well with some BT 4.1 adapters but I'm not sure about that.

virtualabs, 2017-01-12:
I added a few words about Docker containers in the readme. Please try with a VM (I've released one during Hack.lu 2016, see http://bit.ly/hacklu-btlejuice-proxy) and see if it's working better ;)

cryptax, 2017-01-13:
Ok! 
The mega link (for the VM) is no longer available. Can you share it again?

virtualabs, 2017-01-13:
My mistake, I haven't tried it before posting. You can download the VM directly from Mega: https://mega.nz/#!vxU0zIwJ!fMotiX3ARWcwNm67IAtn2ymHKldpVJgo43YkVZrtH4A

cryptax, 2017-01-13:
I am getting plenty of errors with the VM (see attached picture). I'm not very optimistic.
I am a login prompt. What should I use?
PS> I did install the Oracle Extension pack.
[image]

virtualabs, 2017-01-20:
Yep, got the same when a colleague of mine started the VM on his laptop, but it worked pretty well as far as I know. 

Note to self: update the VM with the latest version of BtleJuice

virtualabs, 2017-07-17:
No news since then, issue considered fixed. I'm closing this one.

closed 2017-07-17 by virtualabs

Vaibhav330900, 2018-07-06:
hey,can you tell me the btlejuice login name and password