DigitalSecurity/btlejuice #11: "Connected" event never shows up

idaoudi, 2017-01-18:
Hi,

After a few tries with different BLE devices, I cannot get Btlejuice to work.

I use 2 Debian VM (also tried with a physical and a VM), with 2 distinct Bluetooth 4.0 adapters.
Here is what happened when I try a MitM between TheQuicklock Doorlock and a Nexus 5X mobile device.
The process seems to work correctly, but once I double-click on the target, I can never use the mobile app to connect to the dummy device. The app just keeps searching for the doorlock without finding anything, and I can never see the "Connected" event on Btlejuice web interface.

After a while, a timeout happens, the connection established between the doorlock and the proxy is stopped.
Then, obviously I can use the mobile app correctly to lock and unlock my doorlock, but Btlejuice is not here watching anymore.

Here is the output:

On VM1 (bluetooth service active & hci0 up) : 
> sudo btlejuice-proxy 

    [i] Using interface hci0
    [info] Server listening on port 8000
    [info] Client connected
    Configuring proxy ...
    [status] Acquiring target 20:c3:8f:d9:19:09
    [info] Proxy successfully connected to the real device
    [info] Discovering services and characteristics ...
    [status] Proxy configured and ready to relay !
    [error] Remote device has just disconnected

On VM 2 (bluetooth service inactive & hci0 up):
> sudo btlejuice -u 192.168.5.52 -w

    [i] Using proxy http://192.168.5.52:8000
    [i] Using interface hci0
    2017-01-18T15:40:15.939Z - info: successfully connected to proxy
    2017-01-18T15:40:25.118Z - info: proxy set up and ready to use =)
    2017-01-18T15:40:27.088Z - debug: start advertising
    2017-01-18T15:40:27.110Z - info: BTLE services registered
    2017-01-18T15:40:27.113Z - info: Fixing Bleno handles ...
    2017-01-18T15:41:23.732Z - warn: remote device has disconnected.

Any ideas ?
Thanks !

idaoudi, 2017-01-20:
Ok apparently my bad, I had to spoof the devices BDADDR on VM2. It works ok now

closed 2017-01-20 by idaoudi

saromepsg, 2017-08-09:
hi,
i have made a set up with a Ubunut (Host) machine and a VM (Ubuntu -16.04) on it. Both are configured as per the instruction provided in BTLEJUICE home page. ran the BtleProxy in VM and could see the message "[status] Proxy configured and ready to relay".
Ran the command for web interface in host machine and could see the message " proxy set up and ready to use =)". i am unable to proceed further with data interception and data modification. 

In Host machine, though i have connected a CSR 4.0 BT dongle, it is not listed for hciconfig command. and also how to spoof the BTADDR of target device connected with proxy. Kindly provide the instruction and let me know if anything is missing