DigitalSecurity/btlejuice #12: Error in Discovering device services and Characteristics

malikyaseen, 2017-04-18:
Hi

I am doing research on BLE security. Your work is really helpful. However i am facing following issue

Scenario:
Two Kali Linux machines each having CSR 8510 Bluetooth dongle in a VM environment. VM's successfully connect to eachother but when i double click to connect to BLE device (a simple VR box bluetooth controller) error occurs (detailed error file attached) in discovering its services and characteristics. Moreover is there any way I can simulate BLE advertising device via nrf Connect so that i dont have to use physical ble device?

[error.txt](https://github.com/DigitalSecurity/btlejuice/files/927401/error.txt)

virtualabs, 2017-04-18:
Hi,

Your target device tries to pair with Btlejuice's emulated device, which is only possible on specific Linux kernel versions and associated BlueZ versions (I don't remember the versions but I can find out which ones if required). Btlejuice may handle BTLE "JustWorks" pairing and only this type of pairing. This is a limitation due to the Bleno library Btlejuice's built upon. 

> Moreover is there any way I can simulate BLE advertising device via nrf Connect so that i dont have to use physical ble device?

nrf Connect allows this through its "Advertiser" tab, simply scan nearby BTLE devices and clone them with the included advertiser. It should work on Android.

virtualabs, 2017-07-17:
Your application tries to pair with your device, causing this issue. At the moment, Btlejuice only supports "JustWorks" pairing when installed with a recent version of BlueZ library and a kernel compiled with specific options and modules. That is, it works on a very few setup, but not yours apparently.

malikyaseen, 2017-07-23:
Thanks for the reply
I am pairing using Just Works method and a simulated BLE device using nrf connect (simple temperature sensor) but btlejuice is still unable to show its services and characteristics!!!!

kithack, 2019-08-28:
> Your application tries to pair with your device, causing this issue. At the moment, Btlejuice only supports "JustWorks" pairing when installed with a recent version of BlueZ library and a kernel [...]

Can you name the necessary BlueZ version and the kernel options and modules to make pairing "JustWorks" work?