DigitalSecurity/btlejuice #13: Fitbit Surge - Won't intercept

kapsolas, 2017-04-18:
Hi,

Has anyone tried to use btlejuice with a fitbit surge? I am trying to get the MITM to work and i can't seem to figure out the issue.

My setup is as follows:
2 - separate VMs - They talk to each other successfully
2 - BT adapters. Each assigned to a VM. Model: tbw-106ub

Using the UI, I can select the target device (SURGE).
When I enable BT on my iOS device, it can't connect to the surge anymore. I think its somehow detecting maybe the address is different?

Has anyone had any success with this setup?

virtualabs, 2017-04-18:
Have you disabled the *bluetooth* service in both of your VMs ? This service is well-known to mess up with Btlejuice, as it resets the HCI adapters periodically. 

For more information see https://github.com/sandeepmistry/bleno#linux.

kapsolas, 2017-04-18:
Hmm. I'll have to double check. I don't think I did.

nikoc03, 2017-06-11:
did you manage to do it? I have the same issue. Since the btleproxy connects to the dummy device the application on my phone doesnt seem to be able to pair to the device. Any suggestions?

virtualabs, 2017-07-17:
Have you tried with the latests release of btlejuice ? We improved a lot the discovery process and fixed many issues since then.

sidhub1, 2018-07-23:
I'm also having this problem even after running `sudo service bluetooth stop` on both VMs and downloading the latest version of btlejuice. Please help!

mindfuckup, 2019-05-09:
I can confirm.

My setup:
* 2 up-to-date Kali Linux VMs, VMware Workstation on Windows 10
* A `CSR8510 A10` USB Bluetooth dongle each

Software:
```
# node -v
v8.16.0
# btlejuice -v
1.1.11
```

I can see the spoofed device:
[image]

But I cannot interact with the device:
[image]

Output of `btlejuice-proxy`:
```
# btlejuice-proxy
[info] Server listening on port 8000
[info] Client connected
[warning] client disconnected
[info] Client connected
[i] Stopping current proxy.
Configuring proxy ...
[status] Acquiring target 30:45:11:23:23:55
[info] Proxy successfully connected to the real device
[info] Discovering services and characteristics ...
[status] Proxy configured and ready to relay !
[warning] client disconnected
[info] Client connected
```

Output of `btlejuice`:
```
# btlejuice -u 10.5.23.42 -w
   ___ _   _       __        _
  / __\ |_| | ___  \ \ _   _(_) ___ ___
 /__\// __| |/ _ \  \ \ | | | |/ __/ _ \
/ \/  \ |_| |  __/\_/ / |_| | | (_|  __/
\_____/\__|_|\___\___/ \__,_|_|\___\___|

[i] Using proxy http://10.5.23.42:8000
[i] Using interface hci0
2019-05-09T11:52:54.655Z - info: successfully connected to proxy
2019-05-09T11:53:07.472Z - info: proxy set up and ready to use =)
2019-05-09T11:53:07.564Z - debug: start advertising
2019-05-09T11:53:07.572Z - info: BTLE services registered
2019-05-09T11:53:07.572Z - info: Fixing Bleno handles ...
2019-05-09T11:56:59.057Z - info: dummy: accepted connection from address: 42:66:23:05:13:37
2019-05-09T11:56:59.059Z - info: dummy: disconnected from address: 42:66:23:05:13:37
2019-05-09T11:57:02.147Z - debug: start advertising
2019-05-09T11:57:02.148Z - info: proxy set up and ready to use =)
2019-05-09T11:57:02.232Z - info: BTLE services registered
2019-05-09T11:57:02.234Z - info: Fixing Bleno handles ...
```