DigitalSecurity/btlejuice #14: I'm just getting a blank page.

v4mp1, 2017-05-19:
Hi,

I have set up everything according the readme.

Client is successfully connected to the proxy.

But when I try to connect to the page (browser) I'm just getting a blank page.

I have a "error" on the guest machine. 
After starting the btlejuice-proxy the shell is keep spamming the message: "noble warning: unknown handle 72 disconnected!".

I'm not sure if this is the entire problem or it should working with this "error" aswell.

Thanks!

arunmagesh, 2017-10-05:
Same issue here!

virtualabs, 2017-10-05:
What OS ? Can you copy-paste the output you get from btlejuice, it would be useful to determine what is happening.

whoot, 2018-09-21:
+1 also getting just a blank page :S

whoot, 2018-09-21:
LOL... we are so dumb... 
Just follow the instructions. You/we get the blank page, because you/we requested the Proxy and not btlejuice itself...

Just open http://localhost:8080/ on your host and it will work as expected!

michaelhku2002, 2020-07-16:
@whoot I got the same problem now, and I don;t understand why the btlejuice cannot be requested.

For my case, I'm now running btlejuice and btlejuice-proxy in kali linux installed in raspberry pi. I tried to open the btlejuice web page using firefox in kali linux, but failed. Thanks for your help!

ahmednawazkhan044, 2021-03-16:
Writing following command worked for me.
sudo btlejuice -u IP address -w