DigitalSecurity/btlejuice #15: Error: Cannot find module 'bluetooth-hci-socket'

nikoc03, 2017-06-11:
When I try to run btlejuice-proxy I keep getting the same error:

module.js:487
    throw err;
    ^

Error: Cannot find module 'bluetooth-hci-socket'
    at Function.Module._resolveFilename (module.js:485:15)
    at Function.Module._load (module.js:437:25)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/root/local/lib/node_modules/btlejuice/node_modules/noble/lib/hci-socket/hci.js:6:26)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)

Then I tried to npm install bluetooth-hci-socket but I keep getting:

> bluetooth-hci-socket@0.5.1 install /root/node_modules/bluetooth-hci-socket
> node-gyp rebuild

```
gyp ERR! configure error 
gyp ERR! stack Error: "pre" versions of node cannot be installed, use the --nodedir flag instead
gyp ERR! stack     at install (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:66:16)
gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/node-gyp.js:55:37)
gyp ERR! stack     at getNodeDir (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:77:20)
gyp ERR! stack     at PythonFinder.callback (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:44:7)
gyp ERR! stack     at PythonFinder.<anonymous> (/root/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:470:14)
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:262:7)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:897:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:208:5)
gyp ERR! System Linux 4.9.0-kali4-amd64
gyp ERR! command "/root/local/bin/node" "/root/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/bluetooth-hci-socket
gyp ERR! node -v v9.0.0-pre
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.
npm WARN You are using a pre-release version of node and things may not work as expected

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-hci-socket@0.5.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-06-11T15_48_24_939Z-debug.log
```
Can someone help me please?

virtualabs, 2017-06-12:
Have you tried using sudo to install bluetooth-hci-socket ? What is your OS and version ?

nikoc03, 2017-06-12:
Thank you! Actually I managed to install bluetooth-hci-socket by downgrading the node version to 8 but still I get the following errros when I try to sudo npm install -g btlejuice:

> /usr/local/bin/btlejuice-proxy -> /usr/local/lib/node_modules/btlejuice/bin/cmd_btlejuice_proxy.js
> /usr/local/bin/btlejuice -> /usr/local/lib/node_modules/btlejuice/bin/cmd_btlejuice.js
> 
> > xpc-connection@0.1.4 install /usr/local/lib/node_modules/btlejuice/node_modules/xpc-connection
> > node-gyp rebuild
> 
> gyp ERR! configure error 
> gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/local/lib/node_modules/btlejuice/node_modules/xpc-connection/build'
> gyp ERR! System Linux 4.9.0-kali4-amd64
> gyp ERR! command "/usr/local/bin/node" "/root/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /usr/local/lib/node_modules/btlejuice/node_modules/xpc-connection
> gyp ERR! node -v v8.0.0
> gyp ERR! node-gyp -v v3.6.2
> gyp ERR! not ok 
> 
> > bluetooth-hci-socket@0.5.1 install /usr/local/lib/node_modules/btlejuice/node_modules/bluetooth-hci-socket
> > node-gyp rebuild
> 
> gyp ERR! configure error 
> gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/local/lib/node_modules/btlejuice/node_modules/bluetooth-hci-socket/build'
> gyp ERR! System Linux 4.9.0-kali4-amd64
> gyp ERR! command "/usr/local/bin/node" "/root/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /usr/local/lib/node_modules/btlejuice/node_modules/bluetooth-hci-socket
> gyp ERR! node -v v8.0.0
> gyp ERR! node-gyp -v v3.6.2
> gyp ERR! not ok 
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/btlejuice/node_modules/xpc-connection):
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/btlejuice/node_modules/bluetooth-hci-socket):
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
> 

what could the problem be?

virtualabs, 2017-06-12:
Hope this may help: https://docs.npmjs.com/getting-started/fixing-npm-permissions

nikoc03, 2017-06-12:
Thank you! The thing is that I am already the root user and this is a bit strange since I should have the permission to do this mkdir no?

bik3te, 2017-07-04:
Yes it is a strange error... Another solution is to install the package with the option --unsafe-perm :
$ sudo npm remove -g btlejuice
$ sudo npm install -g btlejuice --unsafe-perm

wikijm, 2018-06-14:
Hi @nikoc03,

Here is a solution to install bluetooth-hci-socket:

```bash
cd ~/
git clone https://github.com/noble/node-bluetooth-hci-socket.git
cd ./node-bluetooth-hci-socket.git
sudo npm install
rm -fR ~/node-bluetooth-hci-socket
```

Works great in this situation:
> OS: Linux kali 4.15.0-kali3-686-pae # SMP Debian 4.15.17-1kali1 (2018-04-25) i686 GNU/Linux
> node: v9.11.1

playaspec, 2018-10-08:
No support for OSX? I'm getting:

specMBP:node-bluetooth-hci-socket gduncan$ sudo npm install

npm ERR! code EBADPLATFORM
npm ERR! notsup Unsupported platform for bluetooth-hci-socket@0.5.2: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm ERR! notsup Valid OS:    linux,android,freebsd,win32
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   darwin
npm ERR! notsup Actual Arch: x64

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/gduncan/.npm/_logs/2018-10-08T22_16_55_505Z-debug.log

 I can't be the only one using this on OS X.

vrdeepakprakash, 2020-03-12:
Install node version manager:

` sudo npm install -g n`

Downgrade node to v 8.9.0 by:

`sudo n 8.9.0`

then

`sudo npm install bluetooth-hci-socket --unsafe-perm`

konacurrents, 2020-05-16:
For osx (mac) I have not been able to get the bluetooth to work, and the bluetooth-hci-socket isn't supported on mac. 

These used to work in 2015 but have since been removed.

Does anyone have a solution to get node-red to talk to my bluetooth devices?
Thanks.
  npm version 6.14.4

aolszowka, 2022-02-12:
@vrdeepakprakash's solution worked for me, as I was doing this in a Dockerfile I also had to install the Python Dependencies, I still got a bunch of errors but that didn't seem to affect the result:

```dockerfile
## Work around bluetooth-hci-socket missing
## https://github.com/DigitalSecurity/btlejuice/issues/15
RUN apt-get -qq install python && \
    npm install -g n && \
    n 8.9.0 && \
    npm install bluetooth-hci-socket --unsafe-perm
```

The whole Dockerfile is here https://github.com/aolszowka/docker-btlejuice/blob/main/btlejuice-proxy/Dockerfile