DigitalSecurity/btlejuice #16: Error: Cannot find module './build/Release/binding.node' on mac os x

nikoc03, 2017-06-12:
Hello everyone, attempting to run btle-proxy on my macbook I get the following:

> module.js:487
>     throw err;
>     ^
> 
> Error: Cannot find module './build/Release/binding.node'
>     at Function.Module._resolveFilename (module.js:485:15)
>     at Function.Module._load (module.js:437:25)
>     at Module.require (module.js:513:17)
>     at require (internal/module.js:11:18)
>     at Object.<anonymous> (/usr/local/lib/node_modules/btlejuice/node_modules/xpc-connection/index.js:3:15)
>     at Module._compile (module.js:569:30)
>     at Object.Module._extensions..js (module.js:580:10)
>     at Module.load (module.js:503:32)
>     at tryModuleLoad (module.js:466:12)
>     at Function.Module._load (module.js:458:3)

Have also created a symbolic sudo ln -s nodejs node in usr/lib
Anyone could help me please?

Iansus, 2017-09-11:
I notice the issue is tagged as solve, but without any solution.
Can the author provide us with the necessary steps?

mounish-p, 2017-09-28:
Try uninstalling the btlejuice and try reinstall with option --unsafe-perm

`sudo npm remove -g btlejuice`
`sudo npm install -g btlejuice --unsafe-perm`

siisko, 2018-03-28:
I had the same issue and **sudo npm install -g btlejuice --unsafe-perm** solved this !

Thank you @mounish-p !

ChiaraSimoni, 2018-07-21:
@siisko @nikoc03 may I ask the exact steps you took to install btlejuice? because I have the same problem but I'm unable to resolve it using this solution..Maybe I did something wrong with the installation procedure

siisko, 2018-07-23:
How did you install btlejuice ? What was the procedure you used ? @ChiaraSimoni

jinggo900, 2019-03-13:
@obihann hey , can i know if you're already know how to fix that issue ? Its the same issue I got. thanks

yesimxev, 2019-06-06:
@jinggo900 @obihann 
Looks like it was just simple missing file / btlejuice is looking for it in the wrong folder.
```
mkdir -p /root/btlejuice/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release
cp /root/btlejuice/node_modules/bluetooth-hci-socket/build/Release/binding.node /root/btlejuice/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release/
```