DigitalSecurity/btlejuice #19: Timeout on service discovery

bubblecup, 2017-06-16:
Hi, thank you so much for this very helpful tool.

I'm able to successfully launch a MITM attack using my existing configuration. However, on one subset of devices I get a timeout on service discovery. IOW, the proxy is able to connect to the device, but not discover its services:

`[info] Proxy successfully connected to the real device`
`[info] Discovering services and characteristics ...`
`[error] discovery timed out, stopping proxy.`

I am able to connect to these devices and browse their services using the Apple Developer Tool Bluetooth Explorer. See a sample screenshot here:

[image]

I was wondering if you'd encountered this issue before and had any idea what could cause it.

Thanks so much.

virtualabs, 2017-06-28:
We eventually managed to reproduce this behavior and released a new version of Btlejuice (1.1.5) that should be fix this problem. Can you give it a try and check if this fixes your problem ?

bubblecup, 2017-06-29:
Hmmm... Well first off, thank you so much for addressing this issue.

When I run the latest version of btlejuice, I get the error: `Error: Cannot find module './build/Release/btim'`.

So, I tried to install btim, but on npm install I get the error:

```
../hci_updown.cpp:6:10: fatal error: 'bluetooth/bluetooth.h' file not found
#include <bluetooth/bluetooth.h>
```

Thanks again for your help.

virtualabs, 2017-06-29:
We've updated the *package.json* file used by npm to add *btim* as a dependency, so if you reinstall btlejuice this would not trigger this error.

You need to install the libbluetooth-dev package if you're using debian or ubuntu, but it's quite weird because this package was already mentionned in btlejuice' setup process as described in the README.

bubblecup, 2017-06-29:
Ahhh ok I come bearing information I should have included originally, sorry...

Install works fine on Rasbian, where I am running the proxy. The machine running the core, where btim install fails with the error message above, is a Mac. I cannot use a Linux VM because VMWare Fusion does not support BTLE drivers on my precise combo of a Mac host and Linux guest...

Note that prior to the dependency on btim, everything was running great though.

Again, thank you so much for your help.

virtualabs, 2017-06-30:
Sounds like btim is not Mac-compatible as it relies on low-level bluetooth libraries. We developed btim to ease BTLE hardware interface management on Linux, but did not give it a try on Mac (as we don't have Mac here). 
Got the point, we will fix it in a next release.

bubblecup, 2017-06-30:
Thank you so much, you guys have been so superb!

Un très gros merci (vous êtes français non?)

bik3te, 2017-07-10:
Just to be sure, did you get a Bad Perm error when installing on MAC ?

Could you try this :
$ sudo npm remove -g btlejuice
$ sudo npm install -g btlejuice --unsafe-perm

I don't believe it will work regarding your error, but when there is a bad permission error, some part of the package is installed...

bubblecup, 2017-07-10:
I did have a Bad Perm error, and I did fix it using the --unsafe-perm option...

bik3te, 2017-07-11:
Yeah but did you remove the package before installing back with --unsafe-perm ?

bubblecup, 2017-07-11:
Oh... Yes I did, although I ran into an NPM issue that I reported here: https://github.com/npm/npm/issues/17503

However, I tried on a fresh OS X install and ran into the same issue with _btim_ as above.

I really do appreciate you getting back to me though, thanks.

bik3te, 2017-07-11:
Sad :/ So give you news if they finally fix the issue ;) And yes we're from France.

virtualabs, 2017-07-12:
We've just released version 1.1.6 that should fix this problem. *btim* is now an optional dependency and is no longer mandatory to run btlejuice. Remove completely btlejuice and btim and then try this new version. 

Hope this will fix your issue.

bubblecup, 2017-07-12:
Hi, thanks! I've uninstalled and reinstalled (on both machines to be exact, the Pi and the Mac), and on the Mac I still get the following error when running the core:

```
module.js:487
    throw err;
    ^

Error: Cannot find module 'btim'
    at Function.Module._resolveFilename (module.js:485:15)
    at Function.Module._load (module.js:437:25)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/btlejuice/bin/cmd_btlejuice.js:20:12)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
```

You still have `var btim = require('btim');` on line 20 of `cmd_btlejuice.js` which I guess is the source of the error.

Thanks again

virtualabs, 2017-07-12:
Allright, I've just fixed btim related issues (once for all I hope) and fixed some minor other bugs by the way. My fellow colleague @narke improved the btim library but haven't publish yet the new version (1.0.2) on Npm, it would available soon (and thus fix the optional dependency install error).

bubblecup, 2017-07-13:
OK, the good news first: It is up and running fine! Thanks for your responsiveness!

Now for the bad stuff: I still get a more verbose variant of the original error, but only for this one particular subset of devices that are all related. Other devices work fine, to be clear. The error I get is:

```
2017-07-13T18:21:57.066Z - info: Fixing Bleno handles ...
/usr/local/lib/node_modules/btlejuice/fake.js:366
  bleno._bindings._gatt._handles = patchedHandles;
                                 ^

TypeError: Cannot set property '_handles' of undefined
    at FakeDevice.fixBlenoHandles (/usr/local/lib/node_modules/btlejuice/fake.js:366:34)
    at Bleno.<anonymous> (/usr/local/lib/node_modules/btlejuice/fake.js:163:17)
    at emitOne (events.js:115:13)
    at Bleno.emit (events.js:210:7)
    at Bleno.onAdvertisingStart (/usr/local/lib/node_modules/btlejuice/node_modules/bleno/lib/bleno.js:159:8)
    at emitOne (events.js:115:13)
    at BlenoBindings.emit (events.js:210:7)
    at BlenoBindings.<anonymous> (/usr/local/lib/node_modules/btlejuice/node_modules/bleno/lib/mac/bindings.js:149:8)
    at emitOne (events.js:115:13)
    at BlenoBindings.emit (events.js:210:7)
```

If you need any additional info about the GATT profile of this device, lmk and I will provide.

Thanks again for all your work.

virtualabs, 2017-07-17:
> OK, the good news first: It is up and running fine! Thanks for your responsiveness!

Glad to hear this :p

It looks like some kind of bad error management in btlejuice, it would be useful to get the following information:
* advertisement and scan data
* complete list of services and characteristics

bubblecup, 2017-07-18:
Hi,

OK so I generated a btsnoop file using the Mac developer utility Packet Logger. I can apply display filters (for advertising packets and services/characteristics) in Wireshark/Tshark but I cannot save the results to a new btsnoop or pcap file (in either Mac or Debian). This appears to be a limitation of Wireshark; I've tried a whole bunch of different things but couldn't find my way to a resolution. So rather than publicly posting my unfiltered data with alllll my bluetooth traffic, which besides being irrelevant is also a privacy concern, I prefer to email it to you if that's ok? I will also send you the display filters you need for this particular device.

I can generate a flat file of results to PDF but I think this way will be more helpful for you.

virtualabs, 2017-07-19:
You can send it by email to virtualabs -youknowwhat- gmail -guesswhatagain- com. You may want to use *mockle*, which is basically a tool able to scan for devices and create a *profile* file (.mock) from a specific device (see README.md).