DigitalSecurity/btlejuice #22: Unable to Connect to proxy error

vaishakbhalli, 2017-07-06:
I get this error all along..."unable to connect to proxy". Do i need to configure it manually?Help and guidance on this matter would be deeply appreciated.
[image]

virtualabs, 2017-07-10:
Did you start the bteljuice proxy manually ?

vaishakbhalli, 2017-07-10:
i didnt configure any proxy externally.

vaishakbhalli, 2017-07-10:
[image]
I tried this again today but didnt turn out to be successful.

virtualabs, 2017-07-10:
You have to run *btlejuice-proxy* on a Rasp Pi equipped with a BT4 adapter
(for instance) and *btlejuice* with *-u* and *-w* on your kali, as you do.
As a matter of fact, if you run no *btlejuice-proxy* on your raspberry, the
tool won't connect because no port is open.

vaishakbhalli, 2017-07-10:
yes i have done that. i have no error on the pi side but i get this error on the kali.

vaishakbhalli, 2017-07-10:
Here's a quick rundown on the usage i have been following so far:
1.on Pi 3
sudo hciconfig
sudo service bluetooth start
sudo hciconfig hci0 up
sudo btlejuice-proxy

2.on Kali
sudo hciconfig 
sudo service bluetooth stop
sudo hciconfig hci1 up(using an external BT4 adapter since my laptop doesn't support BT4)
sudo btlejuice -u IP addr of PI -w -i 1

bik3te, 2017-07-10:
Mmmm, it's strange. How are your rasp and the host connected ? Did you got a Bad perm error when installing btlejuice on the Rasp ?

Could you post a picture of the btlejuice-proxy command execution please ?

vaishakbhalli, 2017-07-11:
i used the following command for the installation on both the Laptop and Pi 3: sudo npm install --unsafe-perm --verbose -g btlejuice. Will post the btlejuice-proxy part picture in a while.

vaishakbhalli, 2017-07-11:
I removed the btlejuice  using this cmd "sudo npm remove -g btlejuice" and when i try to install it from the scratch, i now encounter the following error:
[image]

vaishakbhalli, 2017-07-11:
I removed have removed the Btlejuice on the Laptop but while installing the dependencies , I got the above error.should I proceed with the installation?

bik3te, 2017-07-11:
You can try, it won't get worse but your repos are broken...

vaishakbhalli, 2017-07-11:
So hw do I fix them?

bik3te, 2017-07-11:
vaishakbhalli, I'm sorry but I don't understand how you put yourself in this situation. Anyway, this doesn't seem related to btlejuice. We can't help you for that, better look for solution by googling the error and try to repair it. Or reproducing the install of btlejuice on a fresh ubuntu install. Good luck