DigitalSecurity/btlejuice #25: /usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/crypto.js:60

Trouiller-David, 2017-07-28:
debug: start advertising
info: proxy set up and ready to use =)
info: BTLE services registered
info: Fixing Bleno handles ...
info: dummy: accepted connection from address: 00:1a:7d:da:71:05
/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/crypto.js:60
var output = new Buffer(input.length);

TypeError: Cannot read property 'length' of undefined
    at swap (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/crypto.js:60:32)
    at e (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/crypto.js:37:9)
    at Object.c1 (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/crypto.js:22:9)
    at Smp.handlePairingRandom (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/smp.js:130:12)
    at Smp.onAclStreamData (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/smp.js:58:10)
    at emitTwo (events.js:92:20)
    at emit (events.js:172:7)
    at AclStream.push (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/usr/lib/node_modules/btlejuice/node_modules/bleno/lib/hci-socket/bindings.js:201:21)
    at emitThree (events.js:97:13)

Trouiller-David, 2017-07-29:
Up.

virtualabs, 2017-07-31:
First, it looks like a bug (again) in the newly-added interface management code. We will investigate this.

Second, you cannot get btlejuice working while the bluetooth service is up, as it will definitely interfere with btlejuice. You can disable it permanently (if not used of course) and reboot your machine to see if it has stopped it. Once this service disabled, everything would go fine (well, hopefully).

Trouiller-David, 2017-08-01:
Thank you for your reply #Virtualabs,

I now execute this solution on host  
$ service bluetooth status
$ sudo kill -9 [PID-BLUETOOTHD]
$ sudo btlejuice-proxy  --iface hciX --proxy [IP] --web
And for spoofed mac address i use BDADDR "http://www.petrilopia.net/wordpress/wp-content/uploads/bdaddrtar.bz2"
btlejuice work correctly :), It is really a beautiful project.

Bye.