DigitalSecurity/btlejuice #26: Not able to intercept the data and do the on-the-fly data modification

saromepsg, 2017-08-09:
hi,
i have made a set up with a Ubunut (Host) machine and a VM (Ubuntu -16.04) on it. Both are configured as per the instruction provided in BTLEJUICE home page. ran the BtleProxy in VM and could see the message "[status] Proxy configured and ready to relay".
Ran the command for web interface in host machine and could see the message " proxy set up and ready to use =)". i am unable to proceed further with data interception and data modification.

In Host machine, though i have connected a CSR 4.0 BT dongle, it is not listed for hciconfig command. and also how to spoof the BTADDR of target device connected with proxy. Kindly provide the instruction and let me know if anything is missing

Kay887, 2017-09-14:
Hi there,

Even I am facing the same issue. The proxy shows the Bluetooth device. It says it is ready to relay but for some reason the device does not show up in the app. Could you please help?

MagicScorpion, 2019-07-03:
I am getting the same issue. Any updates on this?