DigitalSecurity/btlejuice #27: Hanging at 'info: Fixing Bleno handles ...'

T08YR4D, 2017-08-10:
Hi, I'm very new to this but I have two VMs set up (Both Ubuntu 16.04) One is using my Mac bluetooth adapter (Proxy) and the other is using an ASUS BT dongle 4.0 (Host) When I run the proxy and the web interface they connect find and I press the shackle on a BT padlock I'm testing and it appears on the list of devices. When I double click on the device, the host hangs on 'info: Fixing Bleno handles ...' an then the device will eventually disconnect. Why is this happening?

adrain1, 2017-10-25:
Have you solved it?

T08YR4D, 2017-10-26:
I think the BLE device I was testing at the time was very secure and it wan't using 'JustWorks'
I don't believe it was an issue with the software.

closed 2017-10-26 by T08YR4D