DigitalSecurity/btlejuice #29: Throw Error - Index out of range

rousez, 2017-10-09:
I was able to setup the proxy on a raspberry pi and use a virtual machine on my Macbook Pro as the host and have the host connect to the proxy.  I then double-click the device (which happens to be a BLE smart lightbulb) in btlejuice web front-end and then the host connecting to the proxy throws the following error:

buffer.js:619
    throw new RangeError('index out of range');
    ^
RangeError: index out of range
    at checkOffset (buffer.js:619:11)
    at Buffer.readUInt16LE (buffer.js:665:5)
    at Gap.onHciLeAdvertisingReport (/home/pi/node_modules/noble/lib/hci-socket/gap.js:100:31)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at Hci.processLeAdvertisingReport (/home/pi/node_modules/noble/lib/hci-socket/hci.js:575:8)
    at Hci.processLeMetaEvent (/home/pi/node_modules/noble/lib/hci-socket/hci.js:534:10)
    at Hci.onSocketData (/home/pi/node_modules/noble/lib/hci-socket/hci.js:429:12)
    at emitOne (events.js:77:13)
    at BluetoothHciSocket.emit (events.js:169:7)

I also have a question regarding the bluetooth adapter versions.  My Macbook Pro's built-in bluetooth adapter is being shared with my linux VM and it is a 5.0 adapter.  Does this matter if the adapter is 5.0 capable?  My raspberry pi has a 4.0 adapter.

virtualabs, 2017-10-17:
Hi rousez,

This is basically due to a newer version of Bleno, a dependency of Btlejuice. The author of this lib made some slight changes in his code that brought this error, so we decided to freeze the version of Bleno Btlejuice relies on.

To get rid of this, please uninstall btlejuice and make sure bleno is completely removed, then install the latest version through npm (v1.1.11 at the time I'm writing this), that should fix this problem.

rousez, 2017-10-19:
@virtualabs Thanks for replying.  I might not be understanding fully - is the recommendation that i should install bleno 1.1.11 through npm?  Will I then have to copy the bleno module folder into btlejuice if I do that?  Looks like at the time I setup btlejuice it has bleno 0.4.0.

rousez, 2017-10-20:
@virtualabs I seemed to have avoided this issue by using my other raspberry pi instead of a VM.  I was also getting an error that stated my hci0 interface did not support Bluetooth Low Energy.  I guess VM's don't work to well with some Bluetooth adapters.  This error is reolved now, the VM seemed to be a show stopper for me.

virtualabs, 2017-10-20:
My first answer was not very clear, but the main idea was to completely reinstall Btlejuice from npm, and make sure it uses the correct version of Bleno as well. 

Using a Raspberry Pi may solve a lot of issues due to the use of a VM, as it may interfere with Bluetooth adapters and cause various errors. I think it'll be interesting to add into the documentation this option, as Raspberry Pis are widely spread and affordable.

closed 2017-10-20 by virtualabs