DigitalSecurity/btlejuice #31: Stopping current proxy just when loading web page

amendezcabrera, 2017-10-27:
Hi, I'm trying to do a MiTM with two of my devices using `btlejuice`, it looks promising, but I am facing a problem, all steps work nicely until the [using web interface](https://github.com/DigitalSecurity/btlejuice/#using-the-web-interface ) section where I can connect but when I  do it the proxy says `Stopping current proxy` without more details, it is when I press the `Select target` button that it throws an error:

> throw error;
>       ^
> 
> Error: Could not start scanning, state is poweredOff (not poweredOn)
>     at Noble.startScanning (/home/user/.npm-global/lib/node_modules/btlejuice/node_modules/noble/lib/noble.js:71:17)
>     at Proxy.scanDevices (/home/user/.npm-global/lib/node_modules/btlejuice/proxy.js:656:9)
>     at Proxy.<anonymous> (/home/user/.npm-global/lib/node_modules/btlejuice/proxy.js:98:12)
>     at emitNone (events.js:86:13)
>     at Socket.emit (events.js:185:7)
>     at /home/user/.npm-global/lib/node_modules/btlejuice/node_modules/socket.io/lib/socket.js:503:12
>     at _combinedTickCallback (internal/process/next_tick.js:73:7)
>     at process._tickCallback (internal/process/next_tick.js:104:9)

It is in the `noble.js file` line 76. I was trying a lot of things, enabling BT, disabling, connecting both devices and more without any success, I would like to know what I'm doing wrong. 
Thank you

Kiina, 2017-11-16:
Bluetooth low energy requires BT 4.0 or higher just in case you haven't figured it out yet

Ern941, 2019-11-05:
Did you resolve it? How?

amendezcabrera, 2019-11-06:
> 
> 
> Did you resolve it? How?

You need a BT 4.0 or higher device in order to use `btlejuice`

Ern941, 2019-11-06:
I understand that i have a problem with Bluetooth on Ubuntu. I can't scan any device and i don't know how to make it works. I use two BT adapters CSR 4.0