DigitalSecurity/btlejuice #35: Specific BLE Device Causes btlejuice to Crash

GamingTechRides, 2018-03-31:
Hello, I am very new to reverse engineering, BLE, node, javascript, linux, programming, etc. I am attempting to MitM a specific BLE device (https://www.homedepot.com/p/Genie-Aladdin-Connect-Smartphone-Enabled-Garage-Door-Controller-to-open-and-monitor-your-door-from-anywhere-ALKT1-R/206268108).

I've already determined the device does not use authentication or encryption and sends its payload to the Master every 30 seconds using a Notify of the Battery Level characteristic. The value it puts in for the Battery Level handle is actually the application-layer data the Master needs.

Here's the problem: when I insert the battery so the device starts advertising- it actually crashes btlejuice. I am running the proxy on a Pi 3 and connecting to it from a different Kali laptop.  Btlejuice runs fine and shows the other BLE devices in the web interface but when I turn on this BLE device- btlejuice crashes with RangeError: index out of range.

I am so new to this I do not even know where to begin troubleshooting. I've attached a screenshot so you can see the error.  It happens every time I turn on the BLE device.  I actually get very similar errors when I attempt to use gattacker, the errors seem to be related to node or noble.  Any help would be greatly appreciated. Thank you.

[image]

virtualabs, 2018-07-08:
Hello,
I've already met this error, and the reason is quite simple: the BLE adapter you are using is not compatible with Noble/Bleno.

Try with a CSR 4.0 USB adapter, and it would work perfectly. This happened to me when I changed my computer and used the integrated BLE adapter (Intel Corporation, BLE version 4.2).

aolszowka, 2022-02-12:
While I know this is old I encountered this on a Rapsberry Pi 3B as well. However the error message was slightly different, or sometimes it would just spin/hang.

As @virtualabs pointed out the BLE Adapter probably isn't compatible, the underlying dependency appears to come from [noble/node-bluetooth-hci-socket](https://github.com/noble/node-bluetooth-hci-socket/blob/15c6eef5731ec4cdbf9ca4af7b2e3b42d9e62fdf/README.md) as of 15c6eef here was the list of supported adapters:

| Name | USB VID | USB PID |
|:---- | :------ | :-------|
| BCM920702 Bluetooth 4.0 | 0x0a5c | 0x21e8 |
| BCM20702A0 Bluetooth 4.0 | 0x19ff | 0x0239 |
| BCM20702A0 Bluetooth 4.0 | 0x0489 | 0xe07a |
| CSR8510 A10 | 0x0a12 | 0x0001 |
| Asus BT-400 | 0x0b05 | 0x17cb |
| Intel Wireless Bluetooth 6235 | 0x8087 | 0x07da |
| Intel Wireless Bluetooth 7260 | 0x8087 | 0x07dc |
| Intel Wireless Bluetooth 7265 | 0x8087 | 0x0a2a |
| Intel Wireless Bluetooth 8265 | 0x8087 | 0x0a2b |
| Belkin BCM20702A0 | 0x050D | 0x065A |

I can report that for me on a Rapsberry Pi 3B that the ASUS BT-400 works.