DigitalSecurity/btlejuice #36: error while running btlejuice-proxy

nishantparhi, 2018-06-30:
root@anonymous:~/btlejuice# btlejuice-proxy 
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'bluetooth-hci-socket'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/root/local/lib/node_modules/btlejuice/node_modules/noble/lib/hci-socket/hci.js:6:26)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
root@anonymous:~/btlejuice#

Vaibhav330900, 2018-07-06:
kindly please have a look on my problem
https://github.com/DigitalSecurity/btlejuice/issues/37

jinggo900, 2019-03-12:
hi nishan, i really hope your kindness if you can reply this message asap. i'm having the same issue like you and can i know if you're already found the problems to solve it. i have been stuck with this problem for almost a week right now :( thank youuuuu