DigitalSecurity/btlejuice #37: Unable to capture the traffic in Btlejuice browser

Vaibhav330900, 2018-07-06:
hello all
My device is connected successfully but I am unable to capture the BLEbulb traffic in Btlejuice web host.
**Vm terminal output is**
xxxx@xxxx-virtual-machine:~$ btlejuice-proxy
[info] Server listening on port 8000
[info] Client connected
[i] Stopping current proxy.
Configuring proxy ...
[status] Acquiring target f8:1d:78:60:7e:42

**Host terminal output is**
xxxxx@xxxxxx:~$ btlejuice -u ipconfig -w
   ___ _   _       __        _          
  / __\ |_| | ___  \ \ _   _(_) ___ ___ 
 /__\// __| |/ _ \  \ \ | | | |/ __/ _ \
/ \/  \ |_| |  __/\_/ / |_| | | (_|  __/
\_____/\__|_|\___\___/ \__,_|_|\___\___|

[i] Using proxy http://xxx.xxx.xxx.xxx:8000
[i] Using interface hci0
2018-07-06T16:25:48.687Z - info: successfully connected to proxy
2018-07-06T16:26:02.592Z - info: proxy set up and ready to use =)

Vaibhav330900, 2018-07-09:
Hey,
I am following this steps-
*Items**
I am using two two BT adapter,smart ble bulb and OS version Ubuntu 16.04 both side host as well as Vm.
Common point -Now first I have the plug both Bt adapter from my laptop port 
**Vm Termianl**
Step 1- hciconfig
Step 2 -Service bluetooth start
Step 3- btlejuice-proxy

**Host terminal**
Step 1- hciconfig
Step 2- Sudo service bluetooth stop
Step 3- sudo hciconfig hci0 up
Step 4 - btlejuice -u <ipconfig vm >  -w

**Output** I have successfully connected my blebulb with btle web page but unable to capture the traffic.

Vaibhav330900, 2018-07-09:
when my host is going to connect with proxy then my btle adapter of host side is automatically going to be disable.

virtualabs, 2018-07-11:
Could be an issue with *btim*, so is this package installed on your machine ? This may be the source of all your troubles (and an item to add in my todo list if this is the case).

Try specifying the HCI interface to use with the *-i hci0* option, it might help.

virtualabs, 2018-07-18:
Did it solve your problem ?

Vaibhav330900, 2018-08-03:
yes Thanks a lot

-- 

-- 
Regards,
Vaibhav Bedi

virtualabs, 2018-08-03:
You're welcome. I'll have a look at this optional library as soon as possible.

I'm closing this issue.

closed 2018-08-03 by virtualabs

farnazforooghifar, 2019-09-18:
Hi. I have the same issue and could not solve it using the answers here. Does anyone know another way that may help?

iqtidar27, 2020-01-08:
Hi,
I am facing the same issue and installed btim in my two Linux machines (16.04). I am unable to capture the traffic and can someone help me to understand the issue? Also about this line "Try specifying the HCI interface to use with the -i hci0 option, it might help."