DigitalSecurity/btlejuice #40: blank page in the web interface

ChiaraSimoni, 2018-07-20:
I'm posting this question again because the other one wasn't answered.
I have ubuntu 16.04 both on host and virtual machine, and was able to install everything correctly, I think, but when I connect to the web interface with the vm ip address I just see a blank page. same thing if I try to connect from the vm. What may be the cause? 
Thanks in advance!

virtualabs, 2018-07-20:
The most probable cause is that you browse the proxy web API instead of the web UI. 
The web UI is available from the host machine on port 8080 (usually at http://localhost:8080/).

ChiaraSimoni, 2018-07-23:
@virtualabs you hit the nail on the head, thanks!
However, even though I was able to connect to the proxy and even select a target from the interface, how come it's still blank? On the proxy it says :
[info] Discovering services and characteristics..
[status] Proxy configured and ready to relay!

on the host: 
Proxy setup and ready to use! =)

I can't really understand why even if i select a target and start the interception it doesn't show anything..
A little bit before the connection I can see a flicker of an image with UUID and characteristics but it disappear almost instantely.

closed 2018-07-24 by ChiaraSimoni