DigitalSecurity/btlejuice #48: Why "BtleJuice Proxy does not work in a Docker container."

erdnussflips, 2019-01-13:
Could anyone explain me, why this not work on in a Docker container?

rubenesco, 2019-11-15:
The reason you can't use both roles of Btlejuice on one machine is how the Linux kernel handles the Bluetooth adapters. It can only support one.

Two docker containers share the same host kernel, so you're going to run into the same issue as a single machine host. With VMs the device handling is done by the VM, not the hypervisor, so you have two kernels

aolszowka, 2022-02-12:
I know this is old; but FWIW based on the comments from @nathanloop the "does not work in a docker container" comment was intended to indicate that you could not run both on the same machine.

For my purposes I have two machines:

1. To run the proxy that is close to the device in question
2. To run the console

Because of this I reasoned that I should be able to create containers and run them on both machines to target their needs. I have done so and created this public repository https://github.com/aolszowka/docker-btlejuice. It seems to be working for the most part for my usage, as of this writing the images aren't optimized, but at least they work.

HUGE thank you for this tool!