DigitalSecurity/btlejuice #49: Help me fix this issue :(   Error: Cannot find module 'bluetooth-hci-socket'

jinggo900, 2019-03-12:
root@anonymous:~/btlejuice# btlejuice-proxy
internal/modules/cjs/loader.js:583
throw err;
^

Error: Cannot find module 'bluetooth-hci-socket'
at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
at Function.Module._load (internal/modules/cjs/loader.js:507:25)
at Module.require (internal/modules/cjs/loader.js:637:17)
at require (internal/modules/cjs/helpers.js:20:18)
at Object. (/root/local/lib/node_modules/btlejuice/node_modules/noble/lib/hci-socket/hci.js:6:26)
at Module._compile (internal/modules/cjs/loader.js:689:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
at Module.load (internal/modules/cjs/loader.js:599:32)
at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
at Function.Module._load (internal/modules/cjs/loader.js:530:3)
root@anonymous:~/btlejuice#

henrikschnor, 2019-04-03:
The module doesn't seem to build properly with newer versions of nodejs, at least this was the issue for me. As a workaround I installed Node.js v8.x and installed btlejuice with the command `sudo npm install --unsafe-perm -g btlejuice`.

shoxxdj, 2019-07-29:
If it helps, 
I had the same issue on archlinux, i've tried all the solutions in the different issues related to this bug and finaly solved with : 

```
sudo  npm i -g @abandonware/bluetooth-hci-socket
cd /usr/lib/node_modules
sudo mv @abandonware/bluetooth-hci-socket .
sudo npm i -g btlejuice --unsafe-perm
```

The basic bluetooth-hci-socket is not maintained I guess ( last version from 2016 )

MoistZuko, 2020-07-27:
> The module doesn't seem to build properly with newer versions of nodejs, at least this was the issue for me. As a workaround I installed Node.js v8.x and installed btlejuice with the command `sudo npm install --unsafe-perm -g btlejuice`.

Using node.js v8.x and --unsafe-perm command works on my Ubuntu20.04.Thanks

Rumpelstiltsquin, 2021-06-23:
> If it helps,
> I had the same issue on archlinux, i've tried all the solutions in the different issues related to this bug and finaly solved with :
>
> ```
> sudo  npm i -g [...]

Thanks, the solution worked for me on Kali