DigitalSecurity/btlejuice #51: Unable to Connect to Advertised Device

mindfuckup, 2019-07-26:
Hi 

I would like to setup BtleJuice but i'm currently not able to get a working setup.

# Tl;Dr

I can select a BLE device in the web interface but the web interface stays empty and I'm also not able to connect to the new advertised device.

# Details

## VM Setup

Used VirtualBox version:

    $ virtualbox  -h
    Oracle VM VirtualBox VM Selector v6.0.10
    [...]

Download latest Kali VM for VirtualBox:

    $ curl -LO https://images.offensive-security.com/virtual-images/kali-linux-2019.2-vbox-amd64.ova

Install VM:

- Import downloaded OVA in VirtualBox
- Network: Bridge to WiFI interface
- Create Snapshot

Start the VM. Change keyboard layount and timezone.

Update the VM:

    # apt-get -y update && apt-get -y upgrade && apt-get -y dist-upgrade && apt-get -y autoremove && apt-get -y autoclean  && apt-get -y clean

Install basic software:

    # apt-get install bluetooth bluez libbluetooth-dev libudev-dev

Install Node Version Manager (nvm):

    # curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.34.0/install.sh | bash

Install and configure node version 8:

    # nvm install 8
    Downloading and installing node v8.16.0...
    Downloading https://nodejs.org/dist/v8.16.0/node-v8.16.0-linux-x64.tar.xz...
    ########################################################################################################################### 100.0%
    Computing checksum with sha256sum
    Checksums matched!
    Now using node v8.16.0 (npm v6.4.1)
    Creating default alias: default -> 8 (-> v8.16.0)
    # nvm use 8
    Now using node v8.16.0 (npm v6.4.1)
    # node --version
    v8.16.0

Install BtleJuice

    # npm install btlejuice

Installed in:

    # ls -l node_modules/btlejuice/
    total 268
    drwx------ 2 root root   4096 Jul 26 10:36 bin
    -rw------- 1 root root   1153 Sep 15  2016 CHANGELOG.md
    -rwx------ 1 root root  10722 Jan 12  2017 core.js
    drwx------ 3 root root   4096 Jul 26 10:36 doc
    -rwx------ 1 root root  17080 Jan 13  2017 fake.js
[... 180 lines cut ...]

The fake device is advertised:

[image]

The web interface does not list any services of the device:

[image]

It's not possible to connect:

[image]

Trying to connect to the spoofed device. However, for example the offical Android App used for this device does not connect and also the nRF application is not able to connect. I assume that nRF should be able to connect, even if the MAC address was not spoofed to the original one.

# Question

Does someone has an idea what I'm doing wrong? I don't see why my setup does not work. 

Note: I have the same issues when I use `gattacker`.

Can I verify somehow if my Bluetooth dongles are "good" ones?

Thanks & best regards,
Emanuel

IHbib, 2019-09-16:
Hey did you find a solution to your issue?

mindfuckup, 2019-09-20:
Hi

No, I was not able to solve the problem. But I got this:

I found a presentation about BLE online: https://nis-summer-school.enisa.europa.eu/2018/cources/IOT/nis-summer-school-damien-cauquil-BLE-workshop.pdf

In there, there is a link to an Ubuntu VM that has btlejuice preinstalled: https://mega.nz/#!nsdxhArR!fGGB2on_JChsmAuT-OORAhDLWdrOgVlu-BRczhFUQXo

I downloaded the VM and I could run btlejuice. It worked better, which means I was able to connect to the advertised "fake" device but it was not very stable. I did not see every packet and I could not all the time manipulate them. So in the end, it was also useless for me.

I don't know why the Ubuntu VMs work but my own Kali setup not. I could not get any more details on that :(.

Some other news: The way to go for me in terms of BLE hacking is to use a BBC Micro:Bit and the btlejack software. This has the disadvantage that you can't perform active man in the middle. You can only sniff traffic (but also from existing connections) and also hijack existing connections (kick out the connected device and connect yourself to the ble device).

Greetz,
Emanuel

IHbib, 2019-10-11:
Hi again and thanks for your suggestion about btlejack, I bought the BBC Microprocessor and will try it next week.

But sadly the VM link is expired, do you perhaps have the VM on your drive and could upload it for me?

mindfuckup, 2019-10-14:
Hi

I do not have the VM anymore. I deleted it b/c as I said, btlejuice or also gattacker did still not work for me in a reliable way on this VM.

trishmapow, 2020-02-01:
Any update on this? There's a similar issue with a different tool I tried earlier https://github.com/securing/gattacker/issues/3.

IHbib, 2020-02-01:
I got it to work by downgrading node.js to version 4.3.2.

mindfuckup, 2020-02-01:
> Any update on this? There's a similar issue with a different tool I tried earlier [securing/gattacker#3](https://github.com/securing/gattacker/issues/3).
> 
> Looks like I might have to go down the Micro:Bit route too.

Hi

I don't have any news. I had the exact same problems when I also tried gattacker. So I'll stick to the Micro::Bit device that works quite well. Not perfect but OKish.

Kunal-Rex, 2021-02-23:
> I got it to work by downgrading node.js to version 4.3.2. Also make sure your bluetooth interfaces both support bluetooth low energy. My notebook didnt support it.

Can you tell me the steps, you did to make it work?

pi3ch, 2022-01-12:
@Kunal-Rex This is my workaround. it comes up and start advertising the dummy device. You can use nrf Connect to connect as see data being intercepted. 

```
# OS Ubuntu 18.4.3 however it should be possible on latest version of Ubuntu as well.
# Remove NPM and NODE that you may have installed via apt get
sudo apt install curl python build-essential bluetooth bluez libbluetooth-dev libudev-dev
# NVM is used to get exact version of node
curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.39.1/install.sh | bash
source ~/.nvm/nvm.sh
# you can change version to any other, 4.3.2
nvm install 8.10.0
npm i -g --unsafe-perm btlejuice
# Sudo to run the remaining command as root
sudo -s 
service bluetooth stop
hciconfig hci0 up
btlejuice -w -u PROXYIP 
# OR
btlejuice-proxy
```

However, I still get `crypto.js:60: TypeError: Cannot read property 'length' of undefined` when a device tries to pair, this seems to be related to underlying library `bleno`: https://github.com/abandonware/bleno/issues/23

iHanxD, 2022-08-30:
@pi3ch I was able to get proxy set up and ready to use. However, the dummy device does not start advertising. Why is that?

[image]