DigitalSecurity/btlejuice #52: Stopping current proxy

farnazforooghifar, 2019-09-18:
Hi. I am trying to use the web interface of btlejuice but as soon as I open it on the virtual machine I have : 
**[i] Stopping current proxy.**

How should I solve this problem? The previous solutions did not help me.

Ern941, 2019-11-04:
I have the same issue, did you resolve it? How?

V33RU, 2020-01-28:
might this OS will helps you ,, but use a proper bluetooth adapter

V33RU, 2020-02-18:
> Hi. I am trying to use the web interface of btlejuice but as soon as I open it on the virtual machine I have :
> **[i] Stopping current proxy.**
> 
> How should I solve this problem? The previous solutions did not help me.

https://github.com/IoT-PTv/IoT-PT 

this OS developed for IoT Pentesting devices check out