DigitalSecurity/btlejuice #54: Can't init device hci0: Connection timed out.

Ern941, 2019-11-04:
Hi, 
I have the following issue:
When i stop the bluetooth service, the hci0 interface goes down and i can't up it using the command "sudo hciconfig hci0 up". I have the error "Can't init device hci0: Connection timed out". How can i resolve it?

Thanks

V33RU, 2020-01-23:
can i see the logs and screenshot of the error might i can help it

V33RU, 2020-02-18:
> Hi,
> I have the following issue:
> When i stop the bluetooth service, the hci0 interface goes down and i can't up it using the command "sudo hciconfig hci0 up". I have the error "Can't init device hci0: Connection timed out". How can i resolve it?
> 
> Thanks

mostly error reason unsupproted adapter or cheap adapter please try to use CSR4.0 a better one might it will resolve the issues