DigitalSecurity/btlejuice #56: Error installing in Ubuntu 16.04

iqtidar27, 2019-12-29:
Hello,
I have installed recent version of node and npm:
[image]

After that trying to install btlejack and got this error:

[image]

[image]

[image]

Would really appreciate any help. Thanks

V33RU, 2019-12-30:
actually issue was nodejs - 8.10.0 
                             npm - 3.5.x
 so it will help you install properly download the 8.10.0 dpkg package and install manually

iqtidar27, 2019-12-31:
Hi @V33RU thanks for your suggestion. Later, i have installed node v8.9.4 and npm 5.6.0. I was able to install btlejuice. But, when i am trying to connect to the proxy  it gives me below error:

[image]

I am using two ubuntu 16 machine and csr4 dongle on both machines. Any suggestion is highly appreciated. 
Note: I am getting similar error for gattacker too.

closed 2020-01-01 by iqtidar27