DigitalSecurity/btlejuice #57: Not able to connect to dummy device

iqtidar27, 2020-01-01:
Hello, 
I am able to run the webserver and show nearby ble devices:

[image]
[image]
Btlejuice created the dummy device for the pulse oximeter but when my oximeter application device was trying to connect to the dummy device but never got connect.
For the Quardioarm it was always trying to make the dummy device but was never successful.

[image]

Does anyone face the same issue? I would really appreciate any suggestion here.

closed 2020-01-08 by iqtidar27

Abdulmalek97, 2020-03-22:
Mind telling us how it was solved?

iqtidar27, 2020-03-22:
It was a trial and error process. I was trying to connect BtleJuice webserver with the Qardioarm android app. So, instead of turning Bluetooth on directly from my android settings, I turned it off initially. When I open the APP it asked me to turn on the Bluetooth, then I turn it on. Finally, it works. Still, it works 60% of the time. Sometimes I had to restart my device to clean up the Bluetooth cash.